MatejKovacic/RasPBX-install #4: Expanding filesystem higs on "Syncing Disks."

jplwoodward, 2022-01-29:
I have tried to follow these instructions on a Pi 4 with raspbx-10-10-2020, but "apt upgrade" hangs at 63%...so I Googled that problem and it was suggested that the filesystem should be expanded first.  So I used raspi-config to do that...and it hangs at "Syncing Disks.".

Anybody know what the problem is?  Is raspbx-10-10-2020 corrupted?

raspbx-04-04-2018 worked fine, but I wanted the latest setup.

jplwoodward, 2022-01-30:
Update...just to be sure I was not just being impatient, I allowed the "apt upgrade" to run overnight...it remained at 63%, convincing me it would never ever complete.  I had also previously allowed the "Syncing Disks" to run overnight, and it never completed.

MatejKovacic, 2022-01-30:
How much space is free on a card?

Is it possible, that SD card is somehow corrupted or faulty? What if you use another SD card (also, another type of it)?

jplwoodward, 2022-01-30:
It is a brand new SanDisk 32GB card.  As I said I cannot expand the filesystem...that hangs.  Here is the df output:

root@raspbx:~# df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        3574000 2490872    905120  74% /
devtmpfs         3900384       0   3900384   0% /dev
tmpfs            4032480       0   4032480   0% /dev/shm
tmpfs            4032480    9600   4022880   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            4032480       0   4032480   0% /sys/fs/cgroup
/dev/mmcblk0p1    258096   55447    202649  22% /boot
tmpfs             806496       0    806496   0% /run/user/0

MatejKovacic, 2022-01-30:
Can you try to download RasPBXagain, verify digital fingerprint (hash), write it on the disk again, then first expand filesystem, reboot and then run apt update & apt upgrade?

jplwoodward, 2022-01-30:
Per your earlier suggestion, I tried the apt-upgrade on a differ sd card (this time 16gb).  It hung at exactly the same place:

Setting up raspberrypi-kernel (1:1.20211201~buster-1) ...
Removing 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack'

Progress: [ 63%] [############################################################################################.......................................................] 

I had already checked the SHA of the original download, and tried expanding the filesystem first (where it hung on syncing disks)but I just now downloaded it again, checked the SHA, flashed it to the new 32GB sd card, added empty ssh file, rebooted, and immediately expanded the filesystem, which again hung:

***@***.***:~# raspi-config

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x35786a18

Device         Boot  Start     End Sectors  Size Id Type
/dev/mmcblk0p1        8192  532481  524290  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532482 7856129 7323648  3.5G 83 Linux

Command (m for help): Partition number (1,2, default 2): 
Partition 2 has been deleted.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (2-4, default 2): First sector (2048-62333951, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (532482-62333951, default 62333951): 
Created a new partition 2 of type 'Linux' and of size 29.5 GiB.
Partition #2 contains a ext4 signature.

Command (m for help): 
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x35786a18

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532481   524290  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532482 62333951 61801470 29.5G 83 Linux

Command (m for help): The partition table has been altered.
Syncing disks.

When I had done this before I left it syncing overnight and it never completed.

Thanks much for your help.

jplwoodward, 2022-01-30:
I just now noticed this in the notes: "Pi 4 with 8GB RAM hangs when running the upgrades. “.  I have 8 gb…so that is the problem.  - John

closed 2022-01-31 by MatejKovacic