MatejKovacic/RasPBX-install #7: Can't hear voice from calling.

guoxiangke, 2022-04-21:
hi, I use RasPBX + E156G, the sms is works very well.
but both calling or receive calling, I can't hear the sounds.

After I plug in the E156G, I got USB0 and 1.
```
root@raspbx:~# ls /dev/ttyUSB
ttyUSB0  ttyUSB1
```
SO, I changed my config to (vi /etc/asterisk/dongle.conf)
```
[dongle0]
audio=/dev/ttyUSB0
data=/dev/ttyUSB1
```
then both send and receive sms works very well, but when calling, I met an error.
```
raspbx*CLI> channel originate dongle/dongle0/+86137****7**9 application MusicOnHold
    -- Called dongle0/+86137****7**9
    -- Dongle/dongle0-0100000009 is making progress
**[2022-04-21 12:56:54] ERROR[18647]: at_response.c:473 at_response_error: [dongle0] AT^DDSETEX failed**
    -- Dongle/dongle0-0100000009 answered
    -- Started music on hold, class 'default', on channel 'Dongle/dongle0-0100000009'
    -- Stopped music on hold on Dongle/dongle0-0100000009
```

MatejKovacic, 2022-04-21:
I guess PIN is disabled, so make sure that VOICE mode is activated on your USB dongle. Also, make sure the firmware of your USB dongle supports VOICE mode - if not, maybe you can flash it with the appropriate firmware?

closed 2022-04-21 by MatejKovacic

guoxiangke, 2022-04-21:
Thanks. fixed, I flash the firmware of the dongle to 11.609.10.02.432 by https://phil.lavin.me.uk/2016/04/enabling-voice-on-a-huawei-e160-atddsetex-failed/.
1.  password(Flash code) from https://a-zgsm.com/freecode/huawei/submit/