MatejKovacic/RasPBX-install #9: Raspberry pi 4 model B hardware revision

workhalid1121, 2023-06-19:
Hello 
I think you are used pi 4.
Which Hardware revision do you use?
Dose raspbx 2020 is still boot on Hardware revision you used?
Can i use asterisk and freepbx on my pi 4 rev 1.4?

Thank you

MatejKovacic, 2023-06-20:
Hmm, I think that hardware revision is not relevant here. If RasPBX boots, everything should be OK. Just update it to the latest version.

Also, I have Raspberry Pi 4 Model B Rev 1.4, cpuinfo revision code is `c03114`.

closed 2023-06-20 by MatejKovacic

workhalid1121, 2023-06-23:
It shows
...........
Start4.elf: is not compatible 
This board requires newer software 
Get the latest software from link

I'm really sorry my rev is 1.5 , revision code is c03115
Any recommendation is appreciated 
Thank you in advance