MatejKovacic/RasPBX-install #10: Raspberry pi 4 model B hardware revision 1.5

workhalid1121, 2023-07-04:
It shows
...........
Start4.elf: is not compatible
This board requires newer software
Get the latest software from link

I'm really sorry my rev is 1.5 , revision code is c03115
Any recommendation is appreciated
Thank you in advance

MatejKovacic, 2023-07-04:
OK, may I suggest a possible solution. Try to get the older board and ssh to the system and run `sudo apt update` and then `sudo apt upgrade`.

I think (however I am not really sure) this would update the OS and it will then boot on your board.

hanyuanzhong, 2024-07-17:
@MatejKovacic hi，After installing raspbx-10-10-2020.zip, boot error, boot should fail in this interface, I don't know how to get ip, then ssh link, and then run sudo apt update and then sudo apt upgrade.
[image]