MatejKovacic/RasPBX-install #15: Can't use outlook mail from microsoft

B0GDAN, 2023-10-28:
Hi all,

Can you share a mail config example with outlook or gmail? I'm trying to set up exim4 with my outlook account and it fails every time. Is there a config step related to SMARTLS that is missing here?

Thanks
Bogdan

MatejKovacic, 2023-10-28:
I guess you should enter `smtp.gmail.com` and port `587` or you can try `465`.

And in `/etc/exim4/passwd.client` you should write something like:
`smtp.gmail.com:my.email:mypassWOrd    # for my.email@gmail.com`

You can also [check this](https://sites.google.com/view/chewkeanho/guides/linux-os/exim4/setup-gmail-send-only).

closed 2023-10-28 by MatejKovacic

B0GDAN, 2023-10-29:
Tried the method there and got to the point where my systems are able to send mails but now I have this:
[image]
How do I clear it?

MatejKovacic, 2023-10-29:
Can you successfully send test mail with a command `send_test_email your.email@server.com`?

Regarding the error in FreePBX, [please check this solution](https://community.freepbx.org/t/freepbx-14-x-dashboard-mail-queue-error-with-exim-installed-and-working/52019).