MatejKovacic/RasPBX-install #16: 能否支持树莓派5(Raspberry PI 5 is supported)

hanyuanzhong, 2024-06-11:


MatejKovacic, 2024-06-11:
Not tested, but I see no reason why it should not work. RPi5 has more RAM and faster CPU, other things are basically the same.

closed 2024-06-11 by MatejKovacic

hanyuanzhong, 2024-06-12:
> 未经测试，但我认为没有理由它不能工作。RPi5 拥有更多 RAM 和更快的 CPU，其他方面基本相同。

I downloaded raspbx-10-10-2020, and after I burned it, it started with an error， I don't know what to do
[image]

MatejKovacic, 2024-06-12:
Ah, I see. I would need to update Raspbian. Unfortunately I don't have much time.

You can try build your own image with the tool published here:
https://www.dslreports.com/forum/r30661088-PBX-FreePBX-for-the-Raspberry-Pi

But be aware, people running that site are real jerks and if you post a question the will likely ban your IP. Just a friendly warning. :)

hanyuanzhong, 2024-06-13:
> 啊，我明白了。我需要更新 Raspbian。不幸的是我没有太多时间。
> 
> 您可以尝试使用此处发布的工具构建自己的图像： [https：//www.dslreports.com/forum/r30661088-PBX-FreePBX-for-the-Raspberry-Pi](https://www.dslreports.com/forum/r30661088-PBX-FreePBX-for-the-Raspberry-Pi)
> 
> 但请注意，运营该网站的人是真正的混蛋，如果你发布问题，他们可能会禁止你的 IP。只是一个友好的警告。:)

Thank you very much for your reply. I will try it. Thanks again