MatejKovacic/RasPBX-install #18: install-dongle error

hanyuanzhong, 2024-06-17:
@MatejKovacic Hello, could you please help me with how to install dongle，I run install-dongle prompt: -bash: install-dongle: command not found
[image]

MatejKovacic, 2024-06-17:
Are you using IMG file from the RasPBX project's website? Or you are using your own OS image?

Try to write `which install-dongle`, it should find it in `/usr/bin/install-dongle`.

If you are using some other operating system, you need to install `install-dongle` manually. Please check this: https://github.com/bg111/asterisk-chan-dongle

hanyuanzhong, 2024-06-18:
Unfortunately, no. As in my 16th issues, I use Raspberry PI 5, so there is no way to use the official image. I installed freepbx according to "https://www.dslreports.com/forum/r30661088-PBX-FreePBX-for-the-Raspberry-Pi", 
so I should manually install "install-dongle", but my installation in "https://github.com/bg111/asterisk-chan-dongle" also failed. Looking at the issues, I found that most people have this problem, but no one has solved the problem
[image]