MatejKovacic/RasPBX-install #19: How can I set up forward to SMS or an email?

saebaryo-xyz, 2024-08-01:
Hi,
First of all, thanks a lot for your tremendous work on explaining from A to Z how to set Raspbx!

I've already set it up with basic set up for calls and sending sms but then I forgot that I needed to receive SMS as well.
Is there a way I can set it up ?
I looked for the options in dongle.conf and there didn't seem to be any lines there.
Your help would be much appreciated!

MatejKovacic, 2024-08-05:
Yes, you need to set up Exim mail server. Check the "Setting up the e-mail system" section.

closed 2024-08-05 by MatejKovacic

saebaryo-xyz, 2024-08-08:
Hey @MatejKovacic seems that I have issues with the Exim4 server.
I am self hosting my Mail server at home, so if I share the conf files and my server settings would be able to double check what's going on?

Thanks a lot!

saebaryo-xyz, 2024-08-08:
Hi again @MatejKovacic really sorry to bother you. I was able to solve my issue with the email server. I totally forgot I was using Sendgrid as Relay so once I changed the credentials it sent the test message like a charm. 

Only issue is to find where to add the recepient email that I missed during the install-dongle set up.

Shall I need to re-run that set up or there is a file where I can add that email?

Thx a lot!!

MatejKovacic, 2024-08-09:
I think you just need to re-run `dpkg-reconfigure exim4-config` and edit `/etc/exim4/passwd.client`.
It is described in a guide.